opendata-stuttgart/sensors-software #964: Particulate matter sensor transmission error

lytton1, 2022-09-10:
The sensor measures the values, but they are no longer displayed on the map 

2 errors are displayed in the device status
OTA Return | connection failed

Data Send Return | connection failed

does anyone have an explanation

pjgueno, 2022-09-10:
Can you send both the config in devices.Sensor.community and in the Sensor itself.
Can you also get the usb Logs ?

lytton1, 2022-09-10:
Hello, I'm not an expert but I hope this is the correct data

 

I haven't changed any of the previous settings either

 

 

 

 

 

 

 

Best regards

Walter

pjgueno, 2022-09-12:
I can't see your pictures...

lytton1, 2022-09-12:
Good Morning,

 

is available again

on this position

lytton1, 2022-10-11:
Hello Pierre-Jean problem is solvedThe cause was the last update from homeassistantThank you for your supportBest regardsWalterVon meinem Telefon gesendet-------- Ursprüngliche Nachricht --------Von: Pierre-Jean Guéno ***@***.***>Datum: Sa., 10. Sept. 2022, 18:13An: opendata-stuttgart/sensors-software ***@***.***>Cc: lytton1 ***@***.***>, Author ***@***.***>Betreff: Re: [opendata-stuttgart/sensors-software] Particulate matter sensor transmission error (Issue #964)
Can you send both the config in devices.Sensor.community and in the Sensor itself.
Can you also get the usb Logs ?

pjgueno, 2022-10-11:
Autoupdate of an home assistant

closed 2022-10-11 by pjgueno