opendata-stuttgart/sensors-software #966: My sensor doesn't work any more

joebar72, 2022-09-20:
In the last 3 months PM2,5 and PM10 data are not collected correctly (absent data). The temperature and humidity fine. What can I do? The restart doesn't work:

https://www.madavi.de/sensor/graph.php?sensor=esp8266-1543953-sds011

Thanks
Red

pjgueno, 2022-09-21:
Please compare the configurations in both the sensor itself and devices.sensor.community. Is the PM sensor the same ?

Then check the connections. A cable could be loose among the four.

joebar72, 2022-09-22:
Thanks
It may be the second one...I only open the box and touched the connections. Now the sensor is working fine.
Regards
Joe

pjgueno, 2022-09-22:
It's almost always the Dupont cables... For my own projects I ve conpletely stopped to use them

pjgueno, 2022-10-11:
Bad connections

closed 2022-10-11 by pjgueno