opendata-stuttgart/sensors-software #971: SDS011 doesn't stop the fan after measurenments

Mr-Hacker11, 2022-11-13:
For a long time, the SDS011 no longer stops the fan after the measurement. My question is, is that how it should be?

pjgueno, 2022-11-18:
Is the sensor actually connected to a WiFi Network ? Are the data displayed on the Map? Please send some pictures proving it. And give Board ID and Sensor ID.
If the sensor is not connected, the fan will turn for 10 minutes after reboot but it actually should stop afterwards.

Mr-Hacker11, 2022-11-20:
Hello, I checked everything. The fan always runs, no matter how it is connected. The sensor is also connected to the WLAN.

If you have any further questions, please contact me.

Best Regards

Simon Vogt

 

Hallo, ich habe schon alles überprüft. Der Sensor ist mit dem WLAN verbunden. Auch die Verbindung zum Sensor ist hergestellt. Auch das wechseln der Drähte bringt nichts.

 

Viele Grüße 

 

Simon Vogt

pjgueno, 2022-11-20:
Please send me the Board ID, the Sensor ID and the link of the map with the Location. I will check.

Mr-Hacker11, 2022-11-20:
Hello, 

the board ID is : 14621585. The Sensor Id is 18-11-16(4d78).

Here you can check the data. https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1 <https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&theme=light&var-chipID=> &theme=light&var-chipID=

Since just now the sensor has been running normally again. The fan stops again.

Best Regards

pjgueno, 2022-11-20:
That is no sensor ID and as I have told you the Sensor was not connected. It is for the past 1 hour. Please check the range of the router and change the PSU. At least 1500 mA. And please edit your comments on GitHub

closed 2022-11-20 by Mr-Hacker11