opendata-stuttgart/sensors-software #973: compilation failed

Troubadix, 2022-12-09:
Hi,

I just bought an Sensirion SEN55 and wanted to try out the sen5x beta firmware.
After a few tries with Arduino IDE and lots of errors a friend told me I have to use VSCode + PlatformIO for this (is this documented anywhere? wasn't able to find)

So I installed Visual Studio Code and PlatformIO Addin, loaded airrohr-firmware.ino and hit the "build" Button.

Only the nodemcuv2_fr was successful build which seems to be due to missing language variables in the intl_*.h files.
I tried to add the missing strings to the intl_de.h but compilation still fails for german. Maybe I did something wrong? Anyway...

I flashed the latest_fr.bin from build directory using esptool to my nodemcu and it boots up fine. When I enter my Wifi settings and reboot the sensor it won't connect to my wifi and still opens the airrohr123456 Wifi for configuration. Wifi password is 100% correct, tried it several times. 
Flashed current stabile firmware for cross-check using airrohr-firmware-flasher.exe and this connects with my wifi as it should be.

So there must be something wrong with my build?

Tried the current master branch for cross-check, which failed to build on all languages :(

What am I doing wrong?

pjgueno, 2022-12-09:
Please contact me through SC forum => pjg.
We need to have a call. I am bored to write 🤣

tk5ep, 2023-04-04:
[Troubadix](https://github.com/Troubadix)
Did you succeed to build something with VS and PIO ?

I'm in the same situation as you. Getting millions of errors.

Thanks for any advices.