opendata-stuttgart/sensors-software #980: Restart nach 1/2 Jahr: Keine Messwerte mehr

LukaGeba, 2023-01-28:
Mein Sensor war seit ca. 6 Monaten nicht mehr aktiv.
Nun habe ich ihn wieder angeschlossen, auch ein FirmwareUpdate gemacht und ich habe trotzdem leider keine Messwerte mehr.
Weder PM noch Temp. Was kann ich tun?

[image]
[image]

pjgueno, 2023-01-28:
Can you check the config. Are you sure it is the right sensors?
Oxidation on the pins ?
What about devices.Sensor.community ?
Please reflash with the firmware flasher.

LukaGeba, 2023-01-28:
Yes, I have oxidation on the Temp.Sensor. But that wouldn't explain why the PM data isn't available either?

I flashed it with "latest.bin", Airrohr Flasher v0.3.4.
I also removed the flash file beforehand and flashed again.

Config: Sensor 1 is SDS011, Sensor 2: DHT22

I am not sure about the correct name / setting for my temp sensor.

pjgueno, 2023-01-29:
Check the config on devices.sensor.community.

pjgueno, 2023-01-29:
Send picture for the temp sensor.