opendata-stuttgart/sensors-software #983: continuous exception (28) with NRZ-2020-133/DE

schattenbewohner, 2023-03-09:
FlashTool: airRohr-firmware-flasher-0.3.4-Windows_amd64_BETA
Binary: latest-de (NRZ-2020-133/DE)
WIFI: local wifi in 1m distance to antenna using 2,4 GHz guest wifi (named GST) without any passwort
OTA: Deactivated

putty log:

```
airRohr: NRZ-2020-133/DE
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to GST
....SNTP synced: Thu Mar  9 19:44:47 2023

.
WiFi connected, IP is: 192.168.1.126

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd30 end: 3fffffc0 offset: 0190
3ffffec0:  00000003 00000000 3ffef4b0 3fffff10
3ffffed0:  3fffff48 00000050 3fffff10 3ffef27f
3ffffee0:  3ffe8939 3ffe8934 3ffef4b0 40218194
3ffffef0:  3ffe892b 3ffe8930 3ffef4b0 4020ab58
3fffff00:  00000040 00000000 feefeffe feefeffe
3fffff10:  70747400 000d0000 80201b44 402013cc
3fffff20:  feefeffe feefeffe feefeffe feefeffe
3fffff30:  feefeffe feefeffe feefeffe feefeffe
3fffff40:  feefeffe feefeffe 00706300 7e01a8c0
3fffff50:  80efeffe 00000000 00000000 feefeffe
3fffff60:  feefeffe feefeffe feefeffe feefeffe
3fffff70:  feefeffe 01004544 feef010d feefeffe
3fffff80:  3ffeea31 feefeffe feefeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe 3ffef6a8
3fffffa0:  3fffdad0 00000000 3ffef668 40223240
3fffffb0:  feefeffe feefeffe 3ffe8690 401006c9
<<<stack<<<
```

pjgueno, 2023-03-09:
Please try with a PW.

pjgueno, 2023-03-09:
And no guest. Microcontroller are no computer.

pjgueno, 2023-03-09:
How did you flash? With PIO ? Did you choose the right env ?

pjgueno, 2023-03-10:
OK you use the beta of the flasher. I should not be an issue.
It is an issue in the board itself.
Which version of the firmware do you select with the flasher?
[image]

Please flash at first without any sensor connected and try to make the board run without any sensor connected.

pjgueno, 2023-03-10:
Also do a full erase with the flasher without attempting to reflash.
The exception error in the ESP are often memory corruptions.

schattenbewohner, 2023-03-10:
thanks for your continous support!
So this is what I did after I read your suggestions:
- removed sensor connector 
- cleared flash
- flashed firmware again (simply latest.bin)
- used "reconfigure" at flash tool to store SSID and password

NodeMCU was ignoring this SSID and tried to access freifunk-disabled.
So I configured MCU via IP to local WIFI (with password) but MCU did not want to connect. (after a while it went to AP mode)
I brought up guest WIFI named GST without password (for testing purposes).
now MCU connects to WIFI but brings exceptions again (**without connected sensors**).

see:

`airRohr: NRZ-2020-133/EN
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to GST
...SNTP synced: Fri Mar 10 22:04:43 2023

twoStageOTAUpdate
output debug text to displays...
.
WiFi connected, IP is: 192.168.1.127

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd30 end: 3fffffc0 offset: 0190
3ffffec0:  00000003 00000000 3ffef4c0 3fffff10
3ffffed0:  3fffff48 00000050 3fffff10 3ffef28f
3ffffee0:  3ffe8939 3ffe8934 3ffef4c0 40218194
3ffffef0:  3ffe892b 3ffe8930 3ffef4c0 4020ab58
3fffff00:  00000040 00000000 feefeffe feefeffe
3fffff10:  70747400 000d0000 80201b44 402013cc
3fffff20:  feefeffe feefeffe feefeffe feefeffe
3fffff30:  feefeffe feefeffe feefeffe feefeffe
3fffff40:  feefeffe feefeffe 00706300 7f01a8c0
3fffff50:  80efeffe 00000000 00000000 feefeffe
3fffff60:  feefeffe feefeffe feefeffe feefeffe
3fffff70:  feefeffe 01004e45 feef010b feefeffe
3fffff80:  3ffeea41 feefeffe feefeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe 3ffef6b8
3fffffa0:  3fffdad0 00000000 3ffef678 40223240
3fffffb0:  feefeffe feefeffe 3ffe8690 401006c9
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------`

This is how my flash tool looks like:
[image]

[image]

pjgueno, 2023-03-10:
Please send me pictures of the board. You are sure it is a nodeMCU v3. Download the normal flasher not the beta.

schattenbewohner, 2023-03-10:
[image]

schattenbewohner, 2023-03-10:
So now: flash cleared using NON-BETA!
downloading latest_de.bin

`airRohr: NRZ-2020-133/DE
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to GST
....SNTP synced: Fri Mar 10 22:24:44 2023

twoStageOTAUpdate
output debug text to displays...
.
WiFi connected, IP is: 192.168.1.127
Starting Webserver... 192.168.1.127

ChipId: 8859298

MAC Id: fcf5c4872ea2
Read SDS...:
Stopping SDS011...
Read DHT...
Send to :
sensor.community
Madavi.de
----
Auto-Update active...`

pjgueno, 2023-03-10:
Ok it is the right uC. Did you find the normal flasher ?

schattenbewohner, 2023-03-10:
so the exceptions are gone but the system still does not connect to WIFI using a password.

schattenbewohner, 2023-03-10:
> Ok it is the right uC. Did you find the normal flasher ?

yes, now I connected sensors to PCB, measurement is working
[image]

next thing I need to address is access to WIFI using password. but maybe not today... :)

pjgueno, 2023-03-10:
It should work now read the full instructions on our web page

pjgueno, 2023-03-10:
We try tomorrow

schattenbewohner, 2023-03-12:
Well, the issue is I am not able to connect to my WIFI6-Wifi @2,4 GHz using password. If I use my cellar wifi router I can connect. But this is ok right now since I put the sensor on a place reaching the "older" wifi.

I will close the issue since the main topic (exceptions) are gone. Seems they were caused by using beta flasher.

Piere-Jean, Thanks a lot for your support!

pjgueno, 2023-03-13:
You're welcome. 
It is sad that my custom flasher now corrupts the firmware. For the PW, your router looks normal, I don't understand.

thepapermen, 2023-05-10:
@pjgueno, built arrohr-firmware-flasher from source (the latest version) and have the same issue after flashing software: no wifi-password after reboot.

thepapermen, 2023-05-12:
@pjgueno, I have built the older version of the firmware flasher from source and it looks like the issue is gone (the actual firmware that I was flashing has always stayed the same). Thanks for your help.

By the way, it's now impossible to build the firmware flasher without modifications to python dependencies, because some packages are gone from PyPi, will probably make a pull request with a fix.

thepapermen, 2023-05-16:
@pjgueno, I have written a Github Action to automatically build the airrohr-firmware-flasher from source on all supported platforms (Linux, OS X, Windows). This action allows to build and download compiled binaries.

[image]

Hopefully, it would make it easier for everyone to contribute to airrohr-firmware-flasher.

Here's the **[pull request.](https://github.com/opendata-stuttgart/airrohr-firmware-flasher/pull/66)**

Phaze-III, 2023-06-17:
> @pjgueno You're welcome. It is sad that my custom flasher now corrupts the firmware.

The PR referenced above hopefully fixes that. The 'Exception (28)' that started this issue most likely was triggered by an empty fs_ssid in the generated config.json. The custom flasher does that when **no** Benutzerdef. Name/Custom name is defined on the Configuration screen. The PR fixes that.

The same error (Exception) can also be triggered on a running sensor by removing the Network Name in the WiFi config setup screen. The exception occurs when the wifi is set up and a 'hostname' is configured from the cfg::fs_ssid. When empty the firmware crashes. This could and probably should be fixed in the firmware (I've a patch ready).

Workaround when using the current beta custom flasher would be to always put something in the Custom Name field although there are then still some issues also addressed in the PR.

Note that the PR doesn't address problems with passwords, sorry about that.

tengelmeier, 2023-07-26:
It looks like my NodeMCU goes into a boot loop even though it is successfully connected to the WLAN.

Airrohr Flasher airRohr-firmware-flasher-0.3.4-MacOS_BETA, 
NodeMCU Amica
Firmware: latest and latest_de_BMx280.bin

No sensors connected

`
Moving Firmware image to old.
Erasing SDK config.
Finished successfully.. Rebooting!

airRohr: NRZ-2020-133/DE
mounting FS…
opened config file…
parsed json…
output debug text to displays…
Connecting to te
….SNTP synced: Wed Jul 26 09:23:29 2023

twoStageOTAUpdate
output debug text to displays…
.
WiFi connected, IP is: 192.168.178.46

——————— CUT HERE FOR EXCEPTION DECODER ———————

Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd30 end: 3fffffc0 offset: 0190
3ffffec0:  00000003 00000000 3ffef4b0 3fffff10  
3ffffed0:  3fffff48 00000050 3fffff10 3ffef27f  
3ffffee0:  3ffe8939 3ffe8934 3ffef4b0 40218194  
3ffffef0:  3ffe892b 3ffe8930 3ffef4b0 4020ab58  
3fffff00:  00000040 00000000 feefeffe feefeffe  
3fffff10:  70747400 000e0000 80201b44 402013cc  
3fffff20:  feefeffe feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe feefeffe  
3fffff40:  feefeffe feefeffe 00706300 2eb2a8c0  
3fffff50:  80efeffe 00000000 00000000 feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe 01004544 feef010d feefeffe  
3fffff80:  3ffeea31 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffef6a8  
3fffffa0:  3fffdad0 00000000 3ffef668 40223240  
3fffffb0:  feefeffe feefeffe 3ffe8690 401006c9  
<<<stack<<<

——————— CUT HERE FOR EXCEPTION DECODER ———————
`

pjgueno, 2023-07-26:
Are some stuff connected to the Board when you flash ?

tengelmeier, 2023-07-26:
Not any more (I tried both with and without). Anyway, it looks like airRohr-firmware-flasher-0.3.4-MacOS_BETA from https://firmware.sensor.community/airrohr/flashing-tool/ is really dated (2021-06-16 01:42).

thepapermen, 2023-07-26:
@tengelmeier, perhaps, you can try the more recent buld of the flasher:

https://github.com/thepapermen/airrohr-firmware-flasher/actions/runs/5003379716

Phaze-III, 2023-07-26:
@tengelmeier Did you use the firmware-flasher to upload a WiFi-configuration and if so did you leave the 'Custom Name' field as is (i.e. no custom name)? 

If that is the case you could try to put some custom name in that field and re-upload the configuration.

tengelmeier, 2023-07-26:
Actually I solved it by flashing with airRohr-firmware-flasher-0.3.2-MacOS and configuring with the (ESP8266) web interface.

@thepapermen Downloading the macOS build results in "quarantined" file (no extension) that macOS offers to open in a text editor. Peeking into it with a hex editor shows it seems to be a binary but as it was downloaded as text it is not executable..

thepapermen, 2023-07-26:
@tengelmeier, thank you. I have added the .dmg extension in the GitHub Action to build to OS X binary. 
https://github.com/thepapermen/airrohr-firmware-flasher/actions/runs/5673268284

tengelmeier, 2023-07-26:
@thepapermen Cool, now it results in a launchable build.

closed 2023-09-06 by ricki-z