opendata-stuttgart/sensors-software #985: WiFi signal stength and BME250 air pressuse on a LCD1602

RuudMR, 2023-04-02:
Hello,
Nice project, but i have a question?
How is it possible to display the value of the WiFi field strength, and signal quality
And from the BME250 sensor the value of the airpressusure on a LCD1602 screen.
With a LCD2004 screen i see the information.  Also with a OLED screen.
But a OLED screen is after a few months burned in with the displayed values and therefor not usable for continue use.
Therefor i have choosen for the LCD1602. (no room for a LCD2004)
But (for now), without the WiFi fieldstenght , signal quality and the BME250 airpressure.
Is there a solution for.  
I can program the ESP8266 if nessesarry? (i am a Arduino teacher) but how?
Greetings from the Netherland.

pjgueno, 2023-04-02:
Hi,
You can add a screen or modify some screens.

Line 3727:
```
display_lines[2] = std::move(tmpl(F("Signal: {v} %"), String(calcWiFiSignalQuality(last_signal_strength))));
```
Line 3705
```
if (p_sensor != "") { display_lines[line_count] = "Pres.: "; display_lines[line_count] += check_display_value(p_value / 100, (-1 / 100.0), 1, 6); display_lines[line_count++] += " hPa"; }
```

These are the lines for the 2004.

Maybe add a screen: uint8_t screens[9];

And then in the switch/case for 1602:
```
case 9:
			display_lines[0] = std::move(tmpl(F("Signal: {v} %"), String(calcWiFiSignalQuality(last_signal_strength))));
                        display_lines[1] = "Pres.: "; display_lines[line_count] += check_display_value(p_value / 100, (-1 / 100.0), 1, 6); display_lines[line_count++] += " hPa"; 
			break;
```

Add also:
```
	if (cfg::display_device_info && cfg::bmx280_read && p_sensor != "") {
		screens[screen_count++] = 8;
	}
```

You will have to flash with Platformio.

RuudMR, 2023-04-02:
pjgueno, nice to hear, i d'know that the source als was free for editting.
I go looking for it and try it out. Thanks.

closed 2023-05-10 by pjgueno