opendata-stuttgart/sensors-software #986: bei jedem reboot geht die WLAN Konfiguration ( Passwort ) verloren und Sensor geht in Konfigurationsmodus

hteibler, 2023-04-02:
bei jedem reboot zb. durch Konfig Änderung oder power-cycle 
geht die WLAN Konfiguration ( Passwort ) verloren und Sensor geht in Konfigurationsmodus

pjgueno, 2023-04-02:
Hi,
How was the firmware flashed. I recommand not to use the beta flasher anymore but the normal one.
The WiFi credentials should be written in the SPIFFS.
Can you send me the serial log ?
Has the sensor always done this ?
It looks like a reboot before writing SPIFFS because of a stack panic. It happened to me with hard modified version of the firmware. Do you use the legit one: latest.bin?

hteibler, 2023-04-02:
hi 
the system is running since years!
today I changed only the API config
since then I can't reboot without loosing WLAN
Firmware: NRZ-2020-133/DE (Nov 29 2020)

pjgueno, 2023-04-02:
>  today I changed only the API config

What did you change exactly?

[image]

Do you know how to read the Serial Log in USB with PIO or the Arduino IDE ?

hteibler, 2023-04-02:
just this:
[image]

hteibler, 2023-04-02:
Do you know how to read the Serial Log in USB with PIO or the Arduino IDE ?   : NO

pjgueno, 2023-04-02:
Have you use some special characters, spaces etc.?
I have never seen any issue while modifying this and I did it a lot.
Can you install the Arduino IDE and then:
[image]
Choose the right port = the one which appears when you plug the sensor in USB(with a data cable!)
[image]
[image]
Try to do everything has usual and send me the text.

pjgueno, 2023-08-06:
Maybe a SPIFFS corruption. Can you erase the config in the interface ?