opendata-stuttgart/sensors-software #988: beta-esp32 branch

Phaze-III, 2023-04-11:
I noticed that the current [beta-esp32](https://github.com/opendata-stuttgart/sensors-software/commits/beta-esp32) branch has [beta-sen5x](https://github.com/opendata-stuttgart/sensors-software/tree/beta-sen5x) as parent but has all Sensirion code removed in commit https://github.com/opendata-stuttgart/sensors-software/commit/82bf9e6c425f17c28ee124527d13544985fd16c6 . It looks like the actual code/files committed  were based on the [beta](https://github.com/opendata-stuttgart/sensors-software/tree/beta) branch. 

I've recreated a branch based on the beta branch so only the specific esp32 modifications are recorded in the git history. The branch can be found on https://github.com/Phaze-III/sensors-software/commits/beta-esp32-only and contains the exact same code as the current beta-esp32 branch but without the beta-sen5x history and without some specific mods for the local environment of @pjgueno . 

I didn't yet create a pull request for this one since that would require a corresponding new branch in the main repository (cleanly branched from the beta branch).

pjgueno, 2023-04-13:
Thank you for tidying my mess...
I made a new branch beta-esp32-v2.
You can PR in.

Phaze-III, 2023-04-15:
Thanks for merging. I also created PR #991 to force the platform_version_esp32 to espressif32@5.2.0 to prevent linking failures when using newer versions and to get the disconnectEventHandler to compile. 

I don't have an ESP32 board yet so can't yet test the firmware but with a little luck other people might now be able to get the code compiled.