opendata-stuttgart/sensors-software #990: wrong ID, sensor comunity

tomkli4837, 2023-04-14:
HI, 

new installation nodemcu v3, flasher generating chip id : 7091141. 
Sensor comunnity when add new device, report `This sensor ID is already registered.` 

pls, help.

pjgueno, 2023-04-15:
Hi,
The sensor has just started to emit:
[https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-7091141](https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-7091141)

There is a probability that some ESP have the same MAC but the probability that 2 persons with such ESP register on devices.sensor.community exactly at the same time should be pretty low...

Have you registered and then delete ?

Please come here:
[https://forum.sensor.community/t/deleted-sensor-mistakenly/1432/27](https://forum.sensor.community/t/deleted-sensor-mistakenly/1432/27)

ricki-z, 2023-04-17:
@tomkli4837 this devices was registered 3 days ago and moment later deleted. The email used to register the device has some similarity to you user name here ...
I've restored this registraton. The device should be shown again in your account at devices.sensor.community.