opendata-stuttgart/sensors-software #995: Support for battery reporting #enhancement

andygodber, 2023-05-10:
Is it possible to include code for rechargeable battery support?
There are ESP8266 devices that a built-in LiPo battery carrier.
Usually A0 is used to read the voltage.

andygodber, 2023-05-10:
Many clones available.
Im using a Dollatek one for something else and thought it would be a great idea.

[https://www.amazon.co.uk/dp/B07MPB2LRT?psc=1&ref=ppx_yo2ov_dt_b_product_details](url)

[image]

[https://www.google.com/search?q=ESP+WROOM+02+Motherboard+Integrated+ESP8266+with+18650+Battery+Slot&rlz=1C1ONGR_en-GBGB974GB974&sxsrf=APwXEdclWAt0bJFZlXAgVDOZLvz-2wOzXQ%3A1683745337675&ei=OepbZOX1KNTZgQal44HwCg&ved=0ahUKEwil9vmVuOv-AhXUbMAKHaVxAK4Q4dUDCA8&uact=5&oq=ESP+WROOM+02+Motherboard+Integrated+ESP8266+with+18650+Battery+Slot&gs_lcp=Cgxnd3Mtd2l6LXNlcnAQA0oECEEYAFAAWABgAGgAcAB4AIABAIgBAJIBAJgBAKABAQ&sclient=gws-wiz-serp](url)

andygodber, 2023-05-10:
One of the WLED Usermods for battery works on this device and gave me the idea - possible make a completely wireless AQI monitor - I will also experiment with a solar charger.

A0 is connected via a 10K resistor to the big +ve pad at the end of the battery.

[image]

https://github.com/Aircoookie/WLED/tree/main/usermods/Battery

pjgueno, 2023-05-10:
Solar would be a good solution.
https://gvidinski.wordpress.com/2021/05/10/solar-powered-nodemcu-sds011-dust-sensor/
Come on the forum. It should be possible to contact Grigori directly. He should be by V3 now.