opendata-stuttgart/sensors-software #996: Sensor displays - 141.6 °C

TheBhim, 2023-07-05:
My sensor keeps displaying the temperature as - 141.6°C. Unplugging and replugging, plus a reebot helps but only for a very short time period, sometimes only seconds.

The sensors are SDS011 & BME280 and the sensor ID is 76256. I am using the power cord and adapter supplied in the kit from Nettigo. No modifications, only setting I changed was enabling data to be send to the Feinstaub app (as far as I remember). 

I asked in the forum as well: https://forum.sensor.community/t/temperature-shows-141-63-c/1808 (might contain additional info).

rexfue, 2023-07-05:
I would say your BME280 is defective and needs to be replaced.

pjgueno, 2023-08-06:
Something new ?

TheBhim, 2023-08-07:
Not really, no. Been away and sensor shows "no data" since many days now for BME280 on devices.sensor.community and values of 0.0 on Feinstaub app. Before this it worked for a short period of time and then did the -141.6 °C again. Haven't had time to restart again yet or find out what options and ways I'd have for a replacement.

svenkubiak, 2023-08-14:
I am having a SPS30 instead of the SDS011, but other than that the exact same behaviour. After Reboot it works for a while, but then SPS30 goes into "nan" and BME280 displays -141.6 °C. 

Already replaced board and BME280, but haven't been able to fix it.

Using Firmware version: NRZ-2020-133/EN (Nov 29 2020) with a clean flash on the new board (esp-8266).

pjgueno, 2023-08-14:
Both the sensors are I2C. Maybe interférence on restart from autoupdate. Please unplug all the Sensors reflash the latest firmware.

svenkubiak, 2023-08-14:
> I am having a SPS30 instead of the SDS011, but other than that the exact same behaviour. After Reboot it works for a while, but then SPS30 goes into "nan" and BME280 displays -141.6 °C.
>
> Already [...]

Update: Did a new test and got solid data for an hour now - which is more than I had before. The only thing I changed was that I moved the sensor from a none-internet-available WiFi to an internet-availabe WiFi. Still odd, because the sensor was running on the none-internet-available WiFi for at least a year, but I got data again.

pjgueno, 2023-08-16:
Just use the flasher and select the latest.bin

svenkubiak, 2023-08-28:
Update: I flashed the device with latest.bin without anything connected and also tried a different esp-8266 and a different BME280. Always ending up in the same issue after a small period of time (hours). The only thing I didn't replace so far was the SPS30 as I don't have a spare part for that. 

Any additional suggestions on how to fix this?

pjgueno, 2023-08-28:
I need pictures of the board and the Connections there are interferences.

svenkubiak, 2023-09-10:
I completely removed the wiring and welded the wires together instead of using Dupont wires. This seems to have fixed the issue.