opendata-stuttgart/sensors-software #998: Remove "aircms.online" "Public air quality monitoring in Russia" from firmware

bf, 2023-07-12:
Hello,

I've just installed my first sensor and airrohr/sensor.community is really amazing. I wanted to enable more sharing of data with other open source projects and noticed that "aircms.online" is one of the recommended options which is officially added to the firmware:

[image]

I notice that this is a russian project for air quality monitoring in cities such as moscov and st petersburg.

Due to the russian invasion of Ukraine and their various well-documented war crimes I think it should be discussed why such a project is added as a recommendation in the firmware. 

I know open source is not political, but there are various sanctions and it should not be possible for a user to accidently share their personal sensor data with a russian entity.

Please let me know what you think. I'm happy to contribute a PR that removes the "aircms.online" link.

ricki-z, 2023-07-17:
This project isn't activated by default. So no one is forced to use it.
The civic community behind this project is normally very sceptic about the government. Are those the people we should sanction? Or wouldn't it be better to support these people, showing that we don't have a problem with 'normal' russian people, just with the actual politics of the government?

Just my 2 cents ...

ituri, 2023-07-18:
> This project isn't activated by default. So no one is forced to use it. The civic community behind this project is normally very sceptic about the government. Are those the people we should [...]

+1

DeeKey, 2023-07-27:
> I notice that this is a Russian project for air quality monitoring in cities such as Moscow and St Peterborough.
That is not correct. Project collects data worldwide and not only at these two cities.
Check the map: https://aircms.online

The reason this API was added previously was intention to avoid unneeded competing and code forking.
Removal of this API might decrease the number of active sensors sending to Sensor.Community API.

I do not know any sanctions applicable to current case.
Thus I don't see any viable reasons to remove aircms.online API from firmware.

pjgueno, 2023-07-27:
Que celui qui n'a jamais péché jette la première pierre...

bf, 2023-07-30:
I think it's a shame how you all play the role of "thoughtful and apolitical friends of Russia" in this discussion. 
Even according to wikipedia, russia is an "authoritarian dictatorship". There is no "civil movements" in authoritarian dictatorships. The IT people who work from within russia on the AirCMS project are most likely the same people who work on russian military projects. Even companies within russia are forced to mobilize people for the war. There is no citizen in the country who is not either passively or actively supporting the genocidal invasion of Ukraine.

Citizens who are against the current dictator are either in prison or have long left the country.

@DeeKey arguing that it's not a russian project is ridiculous, just look at their website, youtube, vkontakte page, etc.
And the argument that there are "no sanctions" does not absolve you from the moral duty to stop promoting russian services.

Generally, I think the pro-russian sentiment shown in this discussion is a violation of the code of conduct at https://codefor.de/grundsaetze/ which says:

"Innerhalb des Netzwerks Code for Germany wollen wir erreichen, dass sich Menschen unabhängig von Geschlecht, sexueller Orientierung, Herkunft, Glauben, Aussehen, Religion oder Alter einbringen und wohlfühlen können. Wir dulden keine sexistischen, rassistischen, homo- oder transfeindlichen, ableistischen oder anderweitig marginalisierte Gruppen diskriminierenden Äußerungen – auch nicht, wenn sie „nur als Witz“ gedacht waren."

russia an authoritarian dictatorship performing a genocidal war on Ukraine with countless documented war crimes 

If you do a deep dive on the people behind this project on the russian side you will most likely find some social media likes for "Z" videos or other content that promotes the russian invasion.

I'll close this issue now.

closed 2023-07-30 by bf