opendata-stuttgart/sensors-software #1000: Please support Sensirion SEN6x sensors

bzed, 2023-07-21:
https://components101.com/sites/default/files/2023-07/SEN6x-datasheet.pdf
Especially interesting: SEN68 – PM, RH&T, VOC & NOx, HCHO

Thanks,

Bernd

bzed, 2023-07-21:
Oh well, Available mid 2024.... Missed that. Sorry!

pjgueno, 2023-08-06:
Hi,
Can you have Sensirion send me a Sensor (when available)? Or you? I did it for the SEN5X.

closed 2023-08-06 by pjgueno

tyeth, 2024-11-22:
There's an eval kit (meant for the SEK SensorBridge) which comes with 3 cables, the jst-ghr to qwiic, dupont header, or rj45 (for SEK Control Center / Sensor Bridge). It was released this month (like a few days ago digikey had 50 in stock).
They've been pushing the release date non stop, and embedded drivers unreleased, all of the models are now Q2 2025 or later. The HCNO was always a later delivery date. I have the eval kit as of this morning, and it has the SEN66 sensor plus cables as mentioned.