opendata-stuttgart/sensors-software #1002: No Data from Sensors?

mikerjensen, 2023-08-25:
Hi, Just set up my first airRohr sensor kit according to the instructions here: https://sensor.community/en/sensors/airrohr/
Unfortunately all is working except no sensor data  from either of the sensors it seems. I checked voltage on the SDS011 board  - - 5.1v (red led pulses about once a second, fan is on).  I have tried both the standard and the beta firmware, same result. I configured  the ESP board before adding the sensors. Pins all look well seated. Sensor is registered on the sensor.community,  ID is 6434359. Wifi data is visible on the Madavi page, but not sensor data. Some debug data below. Any suggestions?
thanks much.
-----
[image]
[image]
[image]

Petarkir2000, 2023-08-25:
Check again wiring!

mikerjensen, 2023-08-25:
Thanks Petarkir2000, I have the latest pre-built wired version, so i haven't done any home-brew wiring. I have double-checked and re-seated all the connectors again (see photo below), but the problem persists.
[image]

mikerjensen, 2023-08-25:
Ah well that is encouraging, I  will check. The plugs can only go in to their sockets in one configuration and in one direction, so I didn't really look at the wiring diagram. I guess the plugs were mis-wired in the first place before they were given to me.

Petarkir2000, 2023-08-25:
See the above scheme!

mikerjensen, 2023-08-25:
You were right! the cable order in the 6pin connector attached to the NodeMCU was reversed. I carved away the plastic on one side of the socket so i could twist the plug 180degrees, and it's working now!

Petarkir2000, 2023-08-25:
Glad this helped you!

pjgueno, 2023-08-25:
Congratulations!

closed 2023-08-25 by pjgueno