opendata-stuttgart/sensors-software #1007: wrong measurement BME280

hemeleerse, 2023-12-26:
I just saw a erratic measurement of the BME280 sensor:
- temperature : - 144,7 °C
- air pressure : 652.26 hPa
- humidity: 100%

How to avoid these ?

carwe, 2023-12-31:
[image]

I noticed this too, but as this was a sensor I built in 2020 and never finaly deployed, I assumed it is broken hardware. To the left you see the values immediately after starting it in 2023 with firmware from 2020, and occasional totally wrong readings. Then it went silent even after power-cycle, so I placed it somewhere with better wifi connection, where it is since then, giving totally wrong readings (for all 3 sensors on the BME280) nearly all the time now. I have not yet checked if it updated the firmware in the meantime (autoupdate is enabled).

In few days a new BME280 will arrive, then I can double-check.

carwe, 2024-01-06:
Replacing the BME280 fixed the issue.