opendata-stuttgart/sensors-software #1011: Senden an apis --> https funktioniert nicht?

Esp2use, 2024-02-16:
[image]

[image]

ohne https funktioniert es wohl

pjgueno, 2024-02-18:
Please send the firmware version. Until a few weeks ago, I was in https. Can you paste the USB logs ?

Esp2use, 2024-02-24:
[image]
[image]

Das hier kam jetzt auch noch hinzu:
[image]

ricki-z, 2024-04-01:
Aktuell sind auf den Servern wieder funktionierende Zertifikate installiert.
In Kürze sollte es dann ein Firmwareupdate geben, das das Problem längerfristiger behebt.

closed 2024-04-01 by ricki-z