opendata-stuttgart/sensors-software #1012: BME 280 suddenly starts delivering weird values

black-puppydog, 2024-02-26:
I've been running this (my first) sensor for about a week now. It' a nettigo pre-flashed kit.

Last night it suddenly started delivering... curious values. See the [grafana](https://api-rrd.madavi.de:3000/grafana/d/q87EBfWGk/temperature-humidity-pressure?var-chipID=esp8266-6787348&var-type=BME280&var-query0=sensors&orgId=1)

[image]

The humidity being stuck at 100% I could understand. there's a clear ramp-up, and in #544 someone pointed out that this can indeed be correct. The abrupt pressure drop I'm unsure about since I have no reference. 

What I *am* sure about is that the temperature reading is off. It's quite a mild spring down here in southern France already, and while last night's rain certainly put a dent in the temperature curve, I am near-certain that it's not -150C on my balcony (although the reading makes me hesitant to verify 😆)

pjgueno, 2024-02-26:
Seems to be a bug. Check the connections and restart

qguv, 2024-06-08:
Similar issue with the same sensor. Happened abruptly, though there is a single datapoint between the last correct value and the new very very cold value. See [this graph](https://api-rrd.madavi.de:3000/grafana/d/q87EBfWGk/temperature-humidity-pressure?var-chipID=esp8266-16748393&orgId=1&viewPanel=5&from=1717192800000&to=1717860628823):

[image]

Note that an identical sensor (placed indoors) does not have this issue, and that they are both running the same firmware:

[image]

[image]