opendata-stuttgart/sensors-software #1021: Keine Zeitsynchronisierung durch Fritzbox ntp

Locutosz, 2024-04-02:
Hallo,
ich habe seit 14 tagen das Problem, daß sich der Luftdatensensor keine Zeitdaten mehr von der Fritzbox holen kann. Der Sensor befindet sich von Anfang an im Gastwlan meiner Fritzbox und hat so über 2 Jahre einwandfrei funktioniert. Auch ein Neustart behebt das Problem nicht. Der Sensor ist einwandfrei im Wlan erreichbar.
Die Firmware ist die NRZ-2021-134-B4.
Angezeigt wird....
NTP Sync | 0
NTP Info | 192.168.179.1 (?) reachable: No
Zeit (UTC) | Thu Jan  1 00:01:58 1970
[image]

Leider gibt es in der Oberfläche keine Möglichkeit einen anderen NTP-Server einzutragen.
Gibt es vielleicht eine neuere Firmware, die das Problem behebt?
Würde mich freuen.

Mit freundlichen Grüßen
Ralf

MichaelDvP, 2024-04-03:
Das Problem ist eher in Deiner Fritzbox zu suchen, vermutlich bekommt sie selber keine Zeit (NTP Einstellung in Heimnetz->Netzwerk->Netzwerkeinstellungen). 
Die Sensorfirmware nutzt `0.pool.ntp.org` und `1.pool.ntp.org`, das wird aber bei der WiFi-Anmeldung per DHCP von der Fritzbox überschrieben. 
Wenn Du die Fritzbox nicht als Zeitserver verwendest, stelle sicher dass der Gastzugang auf UDP-Port 123 zugreifen kann.

Locutosz, 2024-04-03:
Nun hat sich der Sensor ganz aus dem Netzwerk verabschiedet und spannt wieder ein eigenes Netzwerk auf. Leider will dieser nun einen Netzwerkschlüssel haben. Was vielleicht jemand, wie dieser heißt?

ricki-z, 2024-04-03:
wenn nichts geändert wurde sollte das Password 'airrohrcfg' sein.

ricki-z, 2024-04-03:
Zur ursprünglichen Frage, gab es kurz vor dem Problem ein Firmware-Update der Fritzbox? An der Beta-Firmware hat sich seit einigen Monaten nichts geändert. Daher würde ich diese als Grund ausschließen. AVM neigt aber leider dazu, immer mal wieder für die Gastnetze alle Ports außer HTTP(S) und die für Mails benötigten zu sperren. Was echt super ist, da für die Verifizierung von Zertifikaten eine möglichst genaue Zeit gebraucht wird. Wir hatten die Diskussion mit AVM schon einmal, warum NTP vielleicht auch im Gastnetz eine ganz gute Idee sein könnte.

Locutosz, 2024-04-03:
Vielen Dank, das Password hat funktioniert. Ich dachte schon, ich muß wieder ganz von vorn beginnen. Ich hatte vor einiger Zeit die Firmware 7.57 auf die Fritzbox aufgespielt. Das ist aber schon länger als 14 Tage her. Eventuell lief der Sensor ohne Zeitserver eine Weile, bis sich die Zeit soweit verstellt hatte, das daß HTTPS Zerfikat nicht mehr lief. Ich habe mit einem Laptop die Erreichbarkeit von NTP-Servern im Gastnetz überprüft. Diese sind einwandfrei erreichbar und geben auch die Zeit zurück. Dies kann natürlich beim Sensor anders sein.

ricki-z, 2024-04-04:
@Locutosz wir arbeiten gerade an einer neuen Firmware. Diese geht heute in den Beta-Test. Wenn alles gut läuft, sollte sie dann nächste Woche als Stable rausgehen.
(Der Sensor kann nur auf 2.4GHz Kanälen funken. Könnte das der Grund für dafür sein, das der Sensor nicht in das normale Netzwerk kommt?)

Locutosz, 2024-04-04:
Hallo,
danke :) das sind ja positive Nachrichten. Ich habe für 2.4Ghz und 5Ghz jeweils eine andere SSID, so daß ich den jeweiligen Bereich eindeutig zuordnen kann. Der Sensor zeigt ja auch bloß die jeweiligen Router im 2.4Ghz Netzwerk an. ;)

Locutosz, 2024-04-05:
Ich wollte gern nachfragen....ist der Betatest der neuen Firmware nur intern oder kann man sich diese, wenn man den Haken bei lade auch betafirmware setzt, auch jetzt schon erhalten?

Phaze-III, 2024-04-05:
I hope it's OK to reply in English. Since you are using NRZ-2021-134-B4 'Auto Update' won't work. You could however use the airrohr-flasher to flash the current stable version (NRZ-2020-133). That version can be upgraded using 'Auto Update' including the 'Lade Beta Versionen' Haken. You should then get the latest beta (NRZ-2024-134-B6) which should solve some of the mentioned problems.