opendata-stuttgart/sensors-software #1030: Issue with NRZ-2024-135 with BME280 and SDS011 Lolin V3

Perra2v, 2024-04-11:
The values from BME280 get strange even after 24 h after reset  -144.1 c  probably lowest temp

Perra2v, 2024-04-11:
Placed outdoors pipebend style  has been running a long time  with 2020 dec fw and i changed from previous temp/rh sensor
a month and a half ago.  BME280 Shows 581.1 hp instead of 1007 hPa real pressure. works again after reboot then goes down tu the bottom.

Phaze-III, 2024-04-11:
This usually indicates a problem with the wiring (loose connections or corrosion). It could also be an issue with the power supply.

I had this problem a while back and it disappeared after replacing the Dupont-cables with fresh ones.

Perra2v, 2024-04-27:
I made a visit out to my balcony and checked the sensor and eliminator all ok except that the tube had rotated a bit
so it invited some rainwater ... ,-[  secured it in the right position and it worked as before.
Still wondering about the issue with the quick change to -144 i have one indoors without outdoor problems.
But when i use a call from domoticz to read data.json it seems to disturb the registers in the lolin v3 sometimes.

closed 2024-04-27 by Perra2v