opendata-stuttgart/sensors-software #1033: BME280 temperature too high

dr-ni, 2024-04-28:
![Bildschirmfoto zu 2024-04-28 16-40-09](https://github.com/opendata-stuttgart/sensors-
software/assets/68241100/a3615724-087c-4c31-84cf-b368c017615a)

is my sensor broken or is the firmware not measuring correct?
see: https://github.com/arendst/Tasmota/issues/1051#issuecomment-1288423223

dr-ni, 2024-04-28:
[image]

ricki-z, 2024-04-28:
Please check how the BME280 is positioned in the housing. The electronics generate heat that accumulates in the upper area. So try to mount the BME280 as near as possible on an opening.

dr-ni, 2024-04-28:
I'll try next week...
But can the method above help to reduce the temperature?
I'm also searching  the reason off the reason why dht22 humidity always is 99.9%

ricki-z, 2024-04-28:
The high humidity values are a known problem with the DHT22 when used outdoors. Its some kind of saturation.
This is also a reason why we switched to the BME280.
There are many BME280 in our network, especially in Bulgaria. We haven't heard about this problem with our firmware before. As we started using the BME280 we had a similar problem and switched to the recommended settings for the BME280 when used as a "wather station".

dr-ni, 2024-04-28:
ah ok, so this is not helping a long time?

https://forum.arduino.cc/t/arduino-uno-dht22-humidity-99-9/225361/15