opendata-stuttgart/sensors-software #1034: No data upload to opensensemap since 3/16/2024

JimKnopf0, 2024-05-01:
Hi,

my sensor does not upload data to opensensemap since 3/16/2024, sensor.community and madavi.de without HTTPS are fine.
Found some posts indicating this could be a certificate issue. Is there anything, I can do to get this corrected?

[image]

ricki-z, 2024-05-01:
Please update to our newest stable firmware. This firmware should solve this problem.
Unfortunately the version NRZ-2021-134-B4 has a bug that prevents it from updating over the air. So you need to flash the new firmware manually (yous settings should be preserved).

jult, 2024-09-24:
> Please update to our newest stable firmware. This firmware should solve this problem. Unfortunately the version NRZ-2021-134-B4 has a bug that prevents it from updating over the air. So you need to flash the new firmware manually (yous settings should be preserved).

How?

ricki-z, 2024-10-13:
You can find the download link for our firmware flashing tool and instructions on how to flash the firmware on our website https://sensor.community/