opendata-stuttgart/sensors-software #1035: Failed to read (broken?) config

ahaupt, 2024-07-08:
Hi,

I had an older version running for almost 2 years. Then I was hit by #1034 and flashed the current firmware (NRZ-2024-135/DE). After adapting the configuration once after the re-flashing, I cannot change the configuration any longer. There's something wrong:

```
airRohr: NRZ-2024-135/DE
mounting FS...
opened config file...
parsing json: IncompleteInput
failed to load json config
failed to open config file.
No Next PM... serialSDS 9600 8N1
output debug text to displays...
Connecting to Freifunk-disabled
........................................
output debug text to displays...
Starting WiFiManager
AP ID: airRohr-xxxxxxxx
Password: airrohrcfg
scan for wifi networks...
AP Password is: airrohrcfg
Starting Webserver... (IP unset)
```
Somehow the configuration is broken and I cannot fix it. How can I completely delete it? Re-flashing the firmware does not help at all.

Thanks,
Andreas

Phaze-III, 2024-07-09:
If you use the official airrohr firmware flasher (do **not** use the BETA version) there is an option to erase flash before flashing the firmware. That should completely erase any existing and most likely corrupt configurations.

ahaupt, 2024-07-12:
Thanks a lot! Indeed, using the firmware flasher fixed the problem!

Would it be a problem to document the ```esptool``` options it is using to do the actual flashing? The flasher tool is a bit tricky to build as the requirement file refers to Python module versions that are incompatible with at least my Ubuntu-22.04 Python.