opendata-stuttgart/sensors-software #1051: Flash Memory Node Mcu ESP8266

erduino, 2025-02-22:
How much memory flash does the NodeMCU need for the fine dust sensor? 1MB oder 4MB?

erduino, 2025-02-27:
I have a range problem, so I would like to use a NodeMcu with an external antenna, but it only has 1MB of flash. Is that definitely too little?

pjgueno, 2025-02-27:
1M only ? Can I have a picture ?
I have already solder an antenna myself with great success 
[image]

Check example on the internet

erduino, 2025-02-27:
[image]

erduino, 2025-02-27:
Den Lieferanten gibts leider nicht mehr. Aber möglicherweise findet man das Produkt noch in den Untiefen eines Onlineshops